zacharyedwardbull/pycycling #1: device address problems examples

werkgroep-kunstmanen, 2021-02-19:
Hello,
Trying 
tacx_trainer_control_example.py 
and using 
discover_devices_example.py 
to get the device address gives a problem:
The device address in tacx_trainer_control_example.py is EAAA3D1F-6760-4D77-961E-8DDAC1CC9AE which I assume is just an example.

With discover_devices_example.py I get:
C2:9D:1D:49:A9:C7: Tacx Neo 2 01760

which is in a complete different form, and is also not recognized by  the 'tacx trainer' example:

bleak.exc.BleakError: Device with address C2:9D:1D:49:A9:C7 was not found.

What is going wrong here?

Regards,
Rob

zacharyedwardbull, 2021-02-20:
Hi Rob,

The device address is correct, the format is slightly different just due to the way Bluetooth implementations differ between Mac/Windows/Linux.

Please can you try increasing the timeout used when connecting to the turbo trainer?

In `tacx_trainer_control_example.py` change
```python
async with BleakClient(address) as client:
```

to

```python
async with BleakClient(address, timeout=30.0) as client:
```

The timeout is how many seconds to attempt to connect to the device for, you can try with a few different values to see if you have any success.

Also please ensure no other device/software is attempting to connect to the trainer, as I think the turbo can only handle one connection at a time. 

Let me know how it goes,
Zachary

mjcorriere, 2021-02-24:
I can at least throw my 2 cents in here and confirm that on Windows 10, using a device address of the format `EB:14:11:DB:2A:86` lets me successfully connect to my Tacx Neo 2T.

unicornis-pl, 2022-09-22:
I have successfully connected to Wahoo Kickr V5 using address format as above

zacharyedwardbull, 2022-09-22:
> I have successfully connected to Wahoo Kickr V5 using address format as above

That's great, which protocols did you manage to get working with the Kickr @unicornis-pl?

unicornis-pl, 2022-09-24:
I've connected using 
`trainer = CyclingPowerService(client)`
From Wahoo Kickr I get:
CyclingPowerMeasurement(instantaneous_power=106, accumulated_energy=None, pedal_power_balance=None, accumulated_torque=46740, cumulative_wheel_revs=16341, last_wheel_event_time=7986, cumulative_crank_revs=5526, last_crank_event_time=44923, maximum_force_magnitude=None, minimum_force_magnitude=None, maximum_torque_magnitude=None, minimum_torque_magnitude=None, top_dead_spot_angle=None, bottom_dead_spot_angle=None)

unicornis-pl, 2022-09-24:
Similar success connecting to Favero Assioma Duo. Both pedals discovered during scanning, conneting to the left provides all the data:
CyclingPowerMeasurement(instantaneous_power=122, accumulated_energy=None, pedal_power_balance=200, accumulated_torque=None, cumulative_wheel_revs=None, last_wheel_event_time=None, cumulative_crank_revs=15, last_crank_event_time=14928, maximum_force_magnitude=None, minimum_force_magnitude=None, maximum_torque_magnitude=None, minimum_torque_magnitude=None, top_dead_spot_angle=None, bottom_dead_spot_angle=None)

Pedal_power_balance = 200 -> 100% Left
Pedal_power_balance = 0 ->  100% Right
Pedal_power_balance = 100 -> 50/50 Left/Right balance

unicornis-pl, 2022-09-24:
> > I have successfully connected to Wahoo Kickr V5 using address format as above
> 
> That's great, which protocols did you manage to get working with the Kickr @unicornis-pl?

Same result for InPeak crank power sernsor ;-)
`CyclingPowerMeasurement(instantaneous_power=7, accumulated_energy=None, pedal_power_balance=None, accumulated_torque=None, cumulative_wheel_revs=None, last_wheel_event_time=None, cumulative_crank_revs=2, last_crank_event_time=7168, maximum_force_magnitude=None, minimum_force_magnitude=None, maximum_torque_magnitude=None, minimum_torque_magnitude=None, top_dead_spot_angle=None, bottom_dead_spot_angle=None)`

Anything else worth testing?

zacharyedwardbull, 2022-09-24:
> Same result for InPeak crank power sernsor ;-) `CyclingPowerMeasurement(instantaneous_power=7, accumulated_energy=None, pedal_power_balance=None, accumulated_torque=None, cumulative_wheel_revs=None, [...]

No that's great thanks, the reason I ask is because there is an open issue (#13) asking if it already works with the Kickr, and it seems that it does!

closed 2022-12-11 by zacharyedwardbull