zacharyedwardbull/pycycling #2: How to interpret set_basic_resistance

mjcorriere, 2021-02-24:
I'm attempting to write a very simple simulator using pycycling, and my question is about the resistance setting. Can you explain what setting resistance in Newtons means? 

Where is it interpreted that the force is applied? Is this the resistance force applied to the theoretical cyclist? Is it force applied internal to the Tacx unit, say by the magnets on a flywheel?

Thanks!

zacharyedwardbull, 2021-02-24:
Fundamentally, when you cycle two types of forces come into play:
- **Resistive forces**: These are typically rolling resistance, wind resistance, and gravitational resistance. They are forces that would bring you back to a stop if you stop cycling.
- **Inertial forces**: These are the forces that resist changes to your velocity (think [Newton's second law](https://en.wikipedia.org/wiki/Newton%27s_laws_of_motion#Newton's_second_law)). These forces work against you when you accelerate making it hard work to get going (and even harder the heavier you are). However, when you stop pedalling these forces work in your favour, and help counter the resistive forces which slow you down, allowing you to coast along for a while. 

The Tacx NEO is somewhat unique in that it applies both inertial and resistive forces through electromagnets. 

The NEO has 5 different operational modes, which alter the application of these two types of forces. I'll try and explain in a few words what each does:

- **Basic resistance mode**: This is a simple mode, and not useful for many applications. You directly set the resistance force. It simulates a constant flywheel mass which means that this mode is not useful for a simulator (because it therefore applies the incorrect inertial forces for most riders). In pycycling you activate it by using `set_basic_resistance`.
- **Simulation mode**: Here, the trainer computes the resistive force by internally computing a simple physics equation. The equation has parameters which can be configured such as wind speed and track incline. For the full details, please refer to the ANT+ FE-C specification. In this mode, the trainer dynamically adjusts the flywheel mass so it applies the correct inertial force for the rider. To activate this mode, first use `set_user_configuration` then `set_wind_resistance ` and `set_track_resistance` as required. I would strongly recommend using this for any simulator application and this is what is used by Zwift etc.
- **Target power mode (erg mode)**: The trainer adjusts the resistance based on the riders exertion in order to maintain a constant power output. Use `set_target_power ` to activate.
- **Isokinetic mode**: The trainer adjusts the resistance to maintain a constant cadence. See `set_neo_modes` for info on this.
- **Isotonic mode**: This is a special case of basic resistance mode, where the flywheel simulated mass is set to zero rather than a small mass, meaning the inertial forces applied by the trainer are zero. To activate, set bike weight, user weight, incline, drag resistance and rolling resistance to zero and then use `set_basic_resistance` as before.

Let me know if that helps, and if you have any follow up questions. I've been meaning to write proper documentation so I think that just answering common questions is probably a good way to get started!

mjcorriere, 2021-02-24:
Perfect, thanks for clarifying it in so much detail. This is extremely helpful.

In the meantime since opening the issue I signed up for the ANT+ account to get access to the specification to better understand the trainer settings. One of the things that drove me away from using simulation mode was the requirement of the `gear_ratio` parameter. It sounded like somehow the simulator would take into account my gear ratio when determining resistance instead of letting me select a physical gear ratio on my bike. However it looks like this is just used to estimate a cadence when a cadence sensor isn't available (?):

>8.10.2.5 Gear Ratio
The gear ratio (front chain ring teeth:rear wheel cog teeth) entered by the user at the display. This data field may be used
to calculate pedalling cadence by measuring the rear wheel period (cadence). This field allows for a ratio ranging from a
minimum 0.03 (0x01) to a maximum 7.65 (0xFF) between the front and rear gears, in 0.03 increments.

zacharyedwardbull, 2021-02-25:
I'm fairly sure the gear ratio does not affect the resistance applied by the turbo. This is based on my experience with Zwift, where gear changes are done through hardware (physically changing gear), rather than through software. That said, I haven't done any tests so it might be worth confirming yourself (set up simulation mode and then try changing the gear ratio at regular intervals while keeping everything else constant).

closed 2022-04-11 by zacharyedwardbull