zacharyedwardbull/pycycling #3: Problems with installation

KarelPetris, 2021-03-10:
Please how to install it? I'm confused by all the error messages.
I have also tried to install it as administrator, and it has shown me this error:

**`The package setup script has attempted to modify files on your system
that are not within the EasyInstall build area, and has been aborted.

This package cannot be safely installed by EasyInstall, and may not
support alternate installation locations even if you run its setup
script by hand.  Please inform the package's author and the EasyInstall
maintainers to find out if a fix or workaround is available.`**

Thank you very much,
Karel Petříš

zacharyedwardbull, 2021-03-10:
Hi, many thanks for submitting this issue.

Installation should normally be as simple as:
```
pip install pycycling
```

I'm assuming this is what you tried? Installation shouldn't normally require administrator privileges.

Please could you confirm your operating system, and version?

KarelPetris, 2021-03-10:
That I've also tried. 
Error ` Running setup.py install for pythonnet ... error
    ERROR: Command errored out with exit status 1:
     command: 'c:\python39\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\karel\\AppData\\Local\\Temp\\pip-install-vfx_svf4\\pythonnet_084e3b64a00548bea6479a1a7de13252\\setup.py'"'"'; __file__='"'"'C:\\Users\\karel\\AppData\\Local\\Temp\\pip-install-vfx_svf4\\pythonnet_084e3b64a00548bea6479a1a7de13252\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\karel\AppData\Local\Temp\pip-record-t3aic9nf\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\python39\Include\pythonnet'
         cwd: C:\Users\karel\AppData\Local\Temp\pip-install-vfx_svf4\pythonnet_084e3b64a00548bea6479a1a7de13252\
    Complete output (6 lines):
    usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: setup.py --help [cmd1 cmd2 ...]
       or: setup.py --help-commands
       or: setup.py cmd --help

    error: option --single-version-externally-managed not recognized
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'c:\python39\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\karel\\AppData\\Local\\Temp\\pip-install-vfx_svf4\\pythonnet_084e3b64a00548bea6479a1a7de13252\\setup.py'"'"'; __file__='"'"'C:\\Users\\karel\\AppData\\Local\\Temp\\pip-install-vfx_svf4\\pythonnet_084e3b64a00548bea6479a1a7de13252\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\karel\AppData\Local\Temp\pip-record-t3aic9nf\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\python39\Include\pythonnet' Check the logs for full command output.`

zacharyedwardbull, 2021-03-10:
I'm not quite sure what's going wrong here. I'm a macOS user so haven't really tested with Windows (though I think some people have managed to get it working).

Let's leave this issue open, and see if anyone else can chip in with some help. Alternatively, if you manage to figure out what's going wrong, please could you add a note explaining how you fixed it.

Sorry this hasn't been an overly helpful answer, but best of luck.

KarelPetris, 2021-03-11:
I don't know how is it possible, but today I've tried to install it again with "python setup.py develop" and it worked :-)

closed 2021-03-11 by KarelPetris