zacharyedwardbull/pycycling #4: Add FTMS support

zacharyedwardbull, 2021-03-10:
I don't have time myself to implement this (nor do I have an FTMS compatible trainer to test with), but if anyone would like to give it a go, pull requests would be welcome!

The specification can be found here: https://www.bluetooth.com/specifications/specs/

tensorturtle, 2023-01-11:
@zacharyedwardbull I just got a FTMS-compatible trainer (Elite Suito T) so implementing this is certainly on my to-dos!

zacharyedwardbull, 2023-01-13:
That would be fantastic, thanks @tensorturtle! Are the other protocols (CPS and CSCS) working properly with the trainer?

tensorturtle, 2023-01-23:
@zacharyedwardbull I'm almost finished with FTMS support. I can confirm that CPS and CSCS works perfectly with my trainer. A few thoughts / concerns:
1. Compared to CPS and CSCS, FTMS provides much better measurements for speed and cadence because it has a much lower threshold and there is no need to postprocess the data.
2. FTMS is designed to support not just indoor bike trainers but also treadmills and rowing machines, etc. Therefore many of the fields that I have implemented can't be tested and won't be useful in the `pycycling` context.
3. There is a nebulous bluez bug that I can't find any information about, which looks something like: `bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Operation failed with ATT error: 0x80 (Unknown code)`. It happens once a while when `bleak` is writing a GATT characteristic. It's definitely possible to catch the exception and continue. I think I will create a separate issue for this because I don't know if it's a problem with my hardware.

zacharyedwardbull, 2023-01-24:
> 1. Compared to CPS and CSCS, FTMS provides much better measurements for speed and cadence because it has a much lower threshold and there is no need to postprocess the data.

Excellent! Both sound like good things :)

> 2. FTMS is designed to support not just indoor bike trainers but also treadmills and rowing machines, etc. Therefore many of the fields that I have implemented can't be tested and won't be useful in the pycycling context.

That's fine, I guess maybe I should have picked a more general name for the library when I first released it!

> There is a nebulous bluez bug that I can't find any information about, which looks something like: bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Operation failed with ATT error: 0x80 (Unknown [...]

Sorry I'm not able to help with this. https://github.com/hbldh/bleak/issues/174 might be relevant. I find this interesting:

> I am working with a BLE device that will return an application specific ATT error response (between 0x80 - 0x9F, as required by the Bluetooth specification) when a characteristic is written with invalid data.

It sounds like the error might be thrown after writing invalid data, not sure if this is due to a bug in your code, or an issue somewhere else in the stack.

PR #22 (2023-02-04): add FTMS support
Fix #4 This commit implements all 7 characteristics of Fitness Machine Service (FTMS) and demonstrates resistance and power-based control, as well as printing various read/notify characteristics.
closed 2023-02-04 by zacharyedwardbull