zacharyedwardbull/pycycling #5: Support for Bluetooth CYCLING POWER FEATURE Characteristic

ElreboCM, 2021-03-19:
It would be really cool. If you could support the Cycling Power Feature Characteristic as well.

[(https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.cycling_power_vector.xml )]

zacharyedwardbull, 2021-03-19:
Hey, many thanks for creating this issue.

I'm pretty sure we already support the cycling power feature. Try using the `get_cycling_power_feature()` method. 
(https://github.com/zacharyedwardbull/pycycling/blob/8cba37b28432be9d76bf5a4092f0ce453b773919/pycycling/cycling_power_service.py#L113)

Let me know how you get on,
Zachary

ElreboCM, 2021-03-19:
Hi. Yes your are right. 
I will have a closer look. I am testing it with a SRAM QUARQ AXS.

zacharyedwardbull, 2021-03-19:
Did you manage to get pycycling to work with the SRAM QUARQ AXS in the end? Cheers

ElreboCM, 2021-03-20:
The Power Measurement Characteristic works fine.
[image]

However I do not get an out put for the Power Vector Characteristic does not have an output.
Where do you print the output? I could not find it in your code...

`import asyncio
from bleak import BleakClient

from pycycling.cycling_power_service import CyclingPowerService

async def run(address):
    async with BleakClient(address) as client:
        def my_measurement_handler(data):
            print(data)

        await client.is_connected()
        trainer = CyclingPowerService(client)
        trainer.set_cycling_power_measurement_handler(my_measurement_handler)
        #await trainer.enable_cycling_power_measurement_notifications()
        await trainer.enable_cycling_power_vector_notifications()
        await asyncio.sleep(30.0)

if __name__ == "__main__":
    import os

    os.environ["PYTHONASYNCIODEBUG"] = str(1)

   # device_address = "EAAA3D1F-6760-4D77-961E-8DDAC1CC9AED"
    device_address = "01894FCF-5B2B-4468-99C1-A9871D2D4154"
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run(device_address))
`

in your example you are twice subscribing to the Power Measurement Characteristic. I guess that is a mistake?
At least I got an error that I am already subscribed to that characteristic.

ElreboCM, 2021-03-20:
Ah looks like I forgot to set the measurement handler properly. I Will try again when I find some time.

ElreboCM, 2021-03-20:
[image]
That appears to be working 👍

A while ago also tested a Sigeyi AXO. That one was working too!

zacharyedwardbull, 2021-03-20:
Glad you managed to get it to work! 

Improving the documentation is the top of my priority list for this package, hopefully I'll have a chance to do that sometime this summer.

ElreboCM, 2021-03-20:
It is actually quite straightforward to use if you look closely at the code. Nonetheless a bit more documentation won’t harm. 

The only thing that is missing for full cycling power service support is the extreme angles field in the power management characteristic?

ElreboCM, 2021-03-20:
One thing that strikes me however: 

If you look at the torque magnitude array you can spot values above 65000. 
This field is a signed int. Could it be that you are not taking the sign into account when doing the conversion from byte?

ElreboCM, 2021-03-20:
Changed the code in the power vector handler to convert to signed int

` for i in range(byte_offset, len(data), 2):
            element = int.from_bytes(data[i:i + 2], 'little', signed="true")

            if instantaneous_force_array_present:
                instantaneous_force_magnitudes.append(element)
            elif instantaneous_torque_array_present:
                instantaneous_torque_magnitudes.append(element)`

Looking good now 😀

[image]

zacharyedwardbull, 2021-03-21:
Looks like you found a bug! I'll fix this as you have done when I have a free moment.