zacharyedwardbull/pycycling #14: Trouble with the async nature

peteh, 2022-09-06:
Hi, 

I'm having trouble with the asyncio implementation as I'm not too familiar. 

My goal: 
 * read the data from the device
 * publish data via mqtt

Is there are way to just make the example code work with non-async io synchronous code and publish via python's paho-mqtt lib?

zacharyedwardbull, 2022-09-07:
Hi,

I'm not sure you'll be able to make the example work in a synchronous manner - pycycling uses asyncio extensively. It looks like `paho-mqtt` library works with asyncio though, I found an example here [examples/loop_asyncio.py](https://github.com/eclipse/paho.mqtt.python/blob/9782ab81fe7ee3a05e74c7f3e1d03d5611ea4be4/examples/loop_asyncio.py).

Cheers,
Zachary

peteh, 2022-09-07:
Thanks, I'm just not very familiar with programming in these pardigms. 

But maybe I can start with the example you posted. Gotta learn something new every day. 

Thank you very much for now, I'll try to start from there.

closed 2022-09-07 by peteh