zacharyedwardbull/pycycling #18: Implementing pycycling-code with Wahoo Kickr Core

EirikMysen, 2022-11-13:
This GitHub directory was a gift from God to me on my project this fall. I have a Wahoo Kickr Core. Connecting it via cycling_power_service.py was no problem. But since I need the power quite often( I am going to simulate a bike in a game), I have problems. I have used some time to understand asyncio now, but I don't quite understand the coding. 

 async with BleakClient(address) as client:
        def my_measurement_handler(data):
            print(data)

        await client.is_connected()
        
        trainer = CyclingPowerService(client)
        
        trainer.set_cycling_power_measurement_handler(my_measurement_handler)
        await trainer.enable_cycling_power_measurement_notifications()
        await asyncio.sleep(30.0)
        await trainer.disable_cycling_power_measurement_notifications()

The 4 last lines I guess is where I get the problem. Why do the data come out first when asyncio.sleep(30) is running. I thought await meant things had to be done, so the line before with trainer.enable_cycling_power_measurement_notifications() was done.
Might being confusing what I am saying, but anyway, please help me with understanding. I have read about the specific functions/classes on cycling_power_service.py, but I don't quite understand them 100 %

zacharyedwardbull, 2022-11-13:
Hey! 

I think I can understand your confusion, the library uses Python's `asyncio` framework, which is a complicated one to get to grips with.

I think you might benefit from reading up a bit about how asyncio works in Python: the [official documentation](https://docs.python.org/3/library/asyncio-task.html#coroutine) is a good place to start, and I think there are some decent blog posts available as well. 

I think in your case, you are struggling to understand why the code in `my_measurement_handler` runs seemingly at the same time as `await asyncio.sleep(30.0)`? The reason is that with the await keyword, we allow other tasks in the event loop to run while we wait for this one to complete. When the trainer sends a new power measurement, pycycling receives it and then calls the handler we passed to it, in this case `my_measurement_handler`. Finally, after the 30s sleep, execution returns to the second to last line, which stops any further notifications from the turbo trainer.

Hope this helps and let me know if you have follow up questions,
Zach

EirikMysen, 2022-11-22:
So grateful for this help! :) 

1. I have read about asyncio now, so I think I know what happends, but have some questions. Regards to the 30s sleep.  In this case, could we just used a normal synchronous "sleep"-function ? Since there is no other tasks(in the example) that can use this sleep, and the measurement_handler will just run? I guess I am wrong but just want to be sure
2. When you run the program the power output is printed, but which command/function does this?
3. A bit same to question 2.  What I need is the power output in my main function. A part of the  code I have now is 
....
  if game_active:
            
            helicopter_movement += gravity
            #helicopter_movement += Bike.g_watt
            helicopter_rect.centery += helicopter_movement
            screen.blit(helicopter, helicopter_rect)
....
I have turned the main function to asyncio. But how do I get this power value in this function,  by using cycling_power_example?    . Bike.g_watt is just a variable for the example.  

4. And how often can I get the power value, can I get it more often han 1 Hz?

zacharyedwardbull, 2022-11-22:
> 1. I have read about asyncio now, so I think I know what happends, but have some questions. Regards to the 30s sleep.  In this case, could we just used a normal synchronous "sleep"-function ? Since [...]

I think https://stackoverflow.com/questions/56729764/asyncio-sleep-vs-time-sleep-in-python will answer your question better than I can.

> 2. When you run the program the power output is printed, but which command/function does this?

[This line](https://github.com/zacharyedwardbull/pycycling/blob/57184db1a8d39e14775990854e13ae6db3996ba1/examples/cycling_power_service_example.py#L9) prints the CyclingPowerMeasurement object. You can see all of the attributes available to you in the documentation [here](https://zacharybull.com/pycycling/pycycling.cycling_power_service.html#pycycling.cycling_power_service.CyclingPowerMeasurement). In this case you probably want `instantaneous_power` or `accumulated_energy`. In the case of the example you could change the print to `print(data.instantaneous_power)` to just print the instantaneous power measurement.

> 3. A bit same to question 2.  What I need is the power output in my main function. A part of the  code I have now is
>    ....
>    if game_active:
>    ```
>         helicopter_movement += gravity [...]

Probably `data.instantaneous_power`, but hard to say without a lot of context on the rest of your program.

> 
> 4. And how often can I get the power value, can I get it more often han 1 Hz?

Unfortunately I don't think there is a way to increase this through the cycling power service protocol. Uou might find other protocols have the data at higher frequencies - I know for example that the ANT+ FE-C over BLE protocol can do 4Hz, but I don't think this will work with your Wahoo trainer.

closed 2022-12-11 by zacharyedwardbull

EirikMysen, 2022-12-27:
Sorry for not responding to last one..

I am soon done, I am managing the project fine, but It is a bit laggy, but you warned me about this.

You have any idea?

But now all of a sudden ,I can't seem to understand what I have done wrong(I have moved the bike trainer from one place to another), I get all these errors: 
Task exception was never retrieved
future: <Task finished name='Task-15' coro=<run() done, defined at /Users/eirikmysen/pycycling/Egne_tester/Bike.py:13> exception=BleakError('Failed to update the notification status for characteristic 35: Error Domain=CBErrorDomain Code=0 "Unknown error." UserInfo={NSLocalizedDescription=Unknown error.}') created at /Users/eirikmysen/Dropbox/Skole/Fordypningsprosjekt_host_2022/Koding/ball_eksempel.py:117>
source_traceback: Object created at (most recent call last):
  File "/Users/eirikmysen/Dropbox/Skole/Fordypningsprosjekt_host_2022/Koding/ball_eksempel.py", line 141, in <module>
    watt_data()
  File "/Users/eirikmysen/Dropbox/Skole/Fordypningsprosjekt_host_2022/Koding/ball_eksempel.py", line 117, in watt_data
    task=loop.create_task(Bike.run(Bike.device_address))
Traceback (most recent call last):
  File "/Users/eirikmysen/pycycling/Egne_tester/Bike.py", line 29, in run
    await g_trainer.disable_cycling_power_measurement_notifications()
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/pycycling/cycling_power_service.py", line 284, in disable_cycling_power_measurement_notifications
    await self._client.stop_notify(cycling_power_measurement_tx_id)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bleak/__init__.py", line 574, in stop_notify
    await self._backend.stop_notify(char_specifier)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bleak/backends/corebluetooth/client.py", line 380, in stop_notify
    await self._delegate.stop_notifications(characteristic.obj)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bleak/backends/corebluetooth/PeripheralDelegate.py", line 236, in stop_notifications
    await future
bleak.exc.BleakError: Failed to update the notification status for characteristic 35: Error Domain=CBErrorDomain Code=0 "Unknown error." UserInfo={NSLocalizedDescription=Unknown error.}

zacharyedwardbull, 2022-12-27:
Sorry, that's not an error I've seen before, and I couldn't easily find anything about it anywhere else.

EirikMysen, 2022-12-27:
It is like I get the watt and the programs runs fine, and then each 5th second or maybe often this happens. I don't see the logic/pattern here.. don't know if that info helps you more.