zacharyedwardbull/pycycling #20: Rearview radar Magene L508 Unsupported?

ademasi, 2023-01-08:
Hi,

Thanks for your project and the inclusion of the rearview radar support.

I have been trying to get the radar data from the Magene L508.  I can get battery and other information, however the radar information are not coming.

The characteristic id from 
https://github.com/zacharyedwardbull/pycycling/blob/57184db1a8d39e14775990854e13ae6db3996ba1/pycycling/rear_view_radar.py#L22  is unavailable on the Magene.

By running a service discovery with bleak, I found what I think is the good id for this data (8ce5cc03-0a4d-11e9-ab14-d663bd873d93), see below.
``` 
python service_explorer.py  --address FE:1D:F9:DB:75:D9 

2023-01-08 21:50:45,783 __main__ INFO: starting scan...
2023-01-08 21:50:46,219 __main__ INFO: connecting to device...
2023-01-08 21:50:48,559 __main__ INFO: connected
2023-01-08 21:50:48,560 __main__ INFO: [Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 10): Generic Attribute Profile
2023-01-08 21:50:48,560 __main__ INFO: [Service] 8ce5cc01-0a4d-11e9-ab14-d663bd873d93 (Handle: 15): Unknown
2023-01-08 21:50:48,560 __main__ INFO:   [Characteristic] 8ce5cc02-0a4d-11e9-ab14-d663bd873d93 (Handle: 16): Unknown (write-without-response,write,notify)
2023-01-08 21:50:48,936 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 18): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2023-01-08 21:50:49,116 __main__ INFO:   [Characteristic] 8ce5cc04-0a4d-11e9-ab14-d663bd873d93 (Handle: 22): Unknown (read), Value: bytearray(b'0x50')
2023-01-08 21:50:49,116 __main__ INFO:   [Characteristic] 8ce5cc03-0a4d-11e9-ab14-d663bd873d93 (Handle: 19): Unknown (write-without-response,write,notify)
2023-01-08 21:50:49,206 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 21): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2023-01-08 21:50:49,206 __main__ INFO: [Service] 0000180a-0000-1000-8000-00805f9b34fb (Handle: 28): Device Information
2023-01-08 21:50:49,341 __main__ INFO:   [Characteristic] 00002a24-0000-1000-8000-00805f9b34fb (Handle: 31): Model Number String (read), Value: bytearray(b'320')
2023-01-08 21:50:49,431 __main__ INFO:   [Characteristic] 00002a29-0000-1000-8000-00805f9b34fb (Handle: 29): Manufacturer Name String (read), Value: bytearray(b'Qingdao Magene Intelligence Technology Co., Ltd')
2023-01-08 21:50:49,521 __main__ INFO:   [Characteristic] 00002a27-0000-1000-8000-00805f9b34fb (Handle: 35): Hardware Revision String (read), Value: bytearray(b'1')
2023-01-08 21:50:49,611 __main__ INFO:   [Characteristic] 00002a25-0000-1000-8000-00805f9b34fb (Handle: 33): Serial Number String (read), Value: bytearray(b'223201558')
2023-01-08 21:50:49,701 __main__ INFO:   [Characteristic] 00002a26-0000-1000-8000-00805f9b34fb (Handle: 37): Firmware Revision String (read), Value: bytearray(b'0.106')
2023-01-08 21:50:49,791 __main__ INFO:   [Characteristic] 00002a28-0000-1000-8000-00805f9b34fb (Handle: 39): Software Revision String (read), Value: bytearray(b'0.106')
2023-01-08 21:50:49,791 __main__ INFO: [Service] 0000180f-0000-1000-8000-00805f9b34fb (Handle: 24): Battery Service
2023-01-08 21:50:49,881 __main__ INFO:   [Characteristic] 00002a19-0000-1000-8000-00805f9b34fb (Handle: 25): Battery Level (read,notify), Value: bytearray(b'`')
2023-01-08 21:50:50,016 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 27): Client Characteristic Configuration, Value: bytearray(b'\x01\x00')
2023-01-08 21:50:50,016 __main__ INFO: [Service] 0000fe59-0000-1000-8000-00805f9b34fb (Handle: 11): Nordic Semiconductor ASA
2023-01-08 21:50:50,016 __main__ INFO:   [Characteristic] 8ec90003-f315-4f60-9fb8-838830daea50 (Handle: 12): Buttonless DFU (write,indicate)
2023-01-08 21:50:50,106 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 14): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2023-01-08 21:50:50,106 __main__ INFO: disconnecting...
2023-01-08 21:50:52,447 __main__ INFO: disconnected

```
When replacing with the possible good id, I can use the read gatt on it. When I try to subscribe to the notification for this characteristic,  pycycling reply with:

```
python rear_view_radar_example.py
/home/alex/git/pycycling/examples/rear_view_radar_example.py:23: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
/home/alex/git/pycycling/examples/rear_view_radar_example.py:11: FutureWarning: is_connected has been changed to a property. Calling it as an async method will be removed in a future version
  await client.is_connected()
Traceback (most recent call last):
  File "/home/alex/git/pycycling/examples/rear_view_radar_example.py", line 24, in <module>
    loop.run_until_complete(run(device_address))
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/alex/git/pycycling/examples/rear_view_radar_example.py", line 15, in run
    await radar_service.enable_radar_measurement_notifications()
  File "/home/alex/.virtualenvs/radar/lib/python3.10/site-packages/pycycling/rear_view_radar.py", line 60, in enable_radar_measurement_notifications
    await self._client.start_notify(radar_characteristic_id, self._radar_measurement_notification_handler)
  File "/home/alex/.virtualenvs/radar/lib/python3.10/site-packages/bleak/__init__.py", line 649, in start_notify
    await self._backend.start_notify(characteristic, wrapped_callback, **kwargs)
  File "/home/alex/.virtualenvs/radar/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 885, in start_notify
    assert_reply(reply)
  File "/home/alex/.virtualenvs/radar/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 21, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.NotSupported] Operation is not supported

```

If I am using bleak directly, I am not getting anything when cars are coming 

```
python enable_notifications.py  --address  FE:1D:F9:DB:75:D9 -d 8ce5cc02-0a4d-11e9-ab14-d663bd873d93
2023-01-08 21:54:02,944 asyncio  DEBUG: Using selector: EpollSelector
2023-01-08 21:54:02,944 __main__ INFO: starting scan...
2023-01-08 21:54:03,054 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_FE_1D_F9_DB_75_D9): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -52)>}, []]
2023-01-08 21:54:03,057 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_FE_1D_F9_DB_75_D9): ['org.bluez.Device1', {}, ['RSSI']]
2023-01-08 21:54:03,058 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', False)>}, []]
2023-01-08 21:54:03,058 __main__ INFO: connecting to device...
2023-01-08 21:54:03,060 bleak.backends.bluezdbus.client DEBUG: Connecting to device @ FE:1D:F9:DB:75:D9
2023-01-08 21:54:03,067 bleak.backends.bluezdbus.client DEBUG: Connecting to BlueZ path /org/bluez/hci0/dev_FE_1D_F9_DB_75_D9
2023-01-08 21:54:03,313 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_FE_1D_F9_DB_75_D9): ['org.bluez.Device1', {'Connected': <dbus_fast.signature.Variant ('b', True)>}, []]
2023-01-08 21:54:04,282 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_FE_1D_F9_DB_75_D9', {'org.bluez.Battery1': {'Percentage': <dbus_fast.signature.Variant ('y', 95)>, 'Source': <dbus_fast.signature.Variant ('s', GATT Battery Service)>}}]
2023-01-08 21:54:04,483 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_FE_1D_F9_DB_75_D9): ['org.bluez.Device1', {'ServicesResolved': <dbus_fast.signature.Variant ('b', True)>}, []]
2023-01-08 21:54:04,484 __main__ INFO: Connected
2023-01-08 21:54:04,552 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_FE_1D_F9_DB_75_D9/service000f/char0010): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_fast.signature.Variant ('b', True)>}, []]
[... cut ...]
2023-01-08 21:56:04,623 bleak.backends.bluezdbus.client DEBUG: Disconnecting (/org/bluez/hci0/dev_FE_1D_F9_DB_75_D9)
2023-01-08 21:56:04,624 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved (/): ['/org/bluez/hci0/dev_FE_1D_F9_DB_75_D9', ['org.bluez.Battery1']]
```
Maybe @tensorturtle will have more information on this topic.

I can also provide a Wireshark Bluetooth packet capture of the radar messages if needed.

Regards, Alex.

tensorturtle, 2023-01-09:
@ademasi Thanks for submitting this issue. Unfortunately without the device I can't help much. It is odd to me that the Magene L508 would have a different UUID because it is supposedly able to pair with common cycling computers.

ademasi, 2023-01-10:
I think you may be right. It is probable that the UUID is correct (6a4e3203-...), however the current firmware does not support outputting radar data via Bluetooth. The devices compatible with the L508 are all ANT+. 

I have contacted Magene support to get information about the next firmware upgrade.

tensorturtle, 2023-01-11:
@ademasi That makes some sense. Good luck and keep us posted!

ademasi, 2023-01-15:
The answer of Magene is as follows:

`Thank you for writing back. I am sorry that there is no specific timetable for when the radar info will be available via Bluetooth now, and I have feedback your needs again to the radar tail light team.`

So, wait and see, I will keep trying with each firmware update.

Elfsong, 2023-11-23:
I encountered the same problem. 

What I found is that the iOS app "Magene Utility" supports reading the radar information via BLE, so maybe they are using some customized protocol to communicate with the radar?

zacharyedwardbull, 2023-11-23:
@Elfsong I think we have two options:

- Wait until Magene adds support for the official protocol. You could contact them to ask if there is an ETA for when this will be available.
- Implement your own class for the proprietary protocol currently used by device. You could ask Magene for documentation on how this works, see if anyone else has already figured it out, or finally (most difficult) reverse engineer the protocol yourself.

Hope this helps,
Zach

Elfsong, 2023-11-23:
@zacharyedwardbull Thank you for your prompt reply! I will try both.

I tried to capture the Bluetooth packets today and found the app sent two "write" commands to enable the radar data transmission:

[image]

zacharyedwardbull, 2023-11-23:
Nice work! Good luck with the rest of the implementation :)

Elfsong, 2023-11-27:
@zacharyedwardbull @ademasi I've contacted the Magene team, but they've not replied yet. 

I will try to reverse-engineering this protocol then. Since it may be a long run, I will keep you guys updated once I complete it. Thank you!

mamtasiyak, 2024-05-22:
Hey folks,
This is for Magene L508 using ANT+, I am getting data in this format data: **array('B', [81, 255, 12, 1, 105, 26, 215, 13, 128, 21, 136, 40, 5])** where the first 9 values I am not understanding and last 4 are values are for device number, type and transmission type. Last 5 values are always fixed so what this 128 is and what are first 8 values? which ones are range and speed?  if you have any idea what are these could you please help?

tensorturtle, 2024-05-23:
Hi @mamtasiyak thank you for reaching out. I can't help directly, but the definitive resource for you is probably the ANT+ protocol documentation, which I have downloaded and attached for your convenience.

tensorturtle, 2024-05-23:
[D00001665_-_ANT+_Device_Profile_-_Bike_Radar_2.1.pdf](https://github.com/zacharyedwardbull/pycycling/files/15413291/D00001665_-_ANT%2B_Device_Profile_-_Bike_Radar_2.1.pdf)

mamtasiyak, 2024-06-17:
Is it only for me or Magene L508 is not detecting objects for everyone else as well?

I did some experiments with the device in all the modes and It is not detecting objects and even when for once I detected something, the data is incorrect. Is it possible that the device is defected or they are not able to detect objects?

If not Magene then which cheap LiDAR you will recommend which detects objects accurately?

tensorturtle, 2024-06-17:
@mamtasiyak Hi, I can recommend Garmin RVR315. That's the Radar device that the pycycling code was first tested on. LiDAR is a related but different technology (3D and high-resolution but requires lots of postprocessing)

mamtasiyak, 2024-06-17:
Thank you so much @tensorturtle for your prompt response. Why I am asking LiDAR recommendation is because I want to do object classification. Is object classification possible with Radar?

tensorturtle, 2024-06-18:
@mamtasiyak I see. The cycling radar devices (that pycycling supports) doesn't really differentiate between objects and its postprocessing algorithm is hidden from us. The device calculates that internally and returns to us just the distance and speed of each object. LiDAR, which returns 3D point clouds and requires 3D object detection neural network for postprocessing is very much outside the scope of this project.

mamtasiyak, 2024-06-18:
Understood. Thanks @tensorturtle :)

mamtasiyak, 2024-06-21:
Does Garmin RVR315 only detect cars or it detect cycles, humans or any other object?

tensorturtle, 2024-06-24:
> Does Garmin RVR315 only detect cars or it detect cycles, humans or any other object?

Basically it's good at detecting human-or-larger sized objects approaching it at above 10km/h or so.
It's pretty good at detecting motorcycles and fast-approaching cyclists. Bad for regular cyclists and people.

mamtasiyak, 2024-07-01:
Folks my Garmin RVR315 is not able to identify objects coming from left and write, it always says for all the objects approaching from behind even if it is coming from left or right. What do I do?

tensorturtle, 2024-07-02:
@mamtasiyak I don't think this github issue is the appropriate place for discussion regarding specifics of Garmin radar performance-perhaps try official Garmin or related forums?

brinlyau, 2025-03-20:
> [@zacharyedwardbull](https://github.com/zacharyedwardbull) [@ademasi](https://github.com/ademasi) I've contacted the Magene team, but they've not replied yet.
>
> I will try to reverse-engineering this [...]

Did you have any success with this? I'm happy to help you look into this as I want an OSS replacement to their app