zacharyedwardbull/pycycling #31: Sandbox violation while running python setup.py develop

williamrexhardin, 2023-12-21:
> error: Setup script exited with error: SandboxViolation: mkdir('/Users/rex/Library/Caches/com.apple.python/private/var/folders/xs/hbp7kjmd23gbdjbtsgzfw02c0000gn/T/easy_install-xv72vavw', 511) {}
> 
> The package setup script has attempted to modify files on your system
> that are not within the EasyInstall build area, and has been aborted.
> 
> This package cannot be safely installed by EasyInstall, and may not
> support alternate installation locations even if you run its setup
> script by hand.  Please inform the package's author and the EasyInstall
> maintainers to find out if a fix or workaround is available.

Steps to reproduce: `python setup.py develop` 

Running macOS Sonoma 14.1.2 (23B92) && Python 3.9.6 inside a fresh venv.

williamrexhardin, 2023-12-21:
Running `pip install bleak` as a precursor to running setup.py resolved the issue. For the record I am on Apple Silicon / M2, perhaps it's just a transitive dependency issue.

zacharyedwardbull, 2023-12-21:
Hey, thanks for submitting this issue.

Please try installing Bleak on its own and see if it fails with the same error. If so, you could file a bug with Bleak and we can close this one.

Cheers,
Zach