zacharyedwardbull/pycycling #33: fitness_machine_service.py: Fitness machine status characteristic id is incorrect

coniesco, 2024-01-01:
Thank you for this awesome library! I'm trying to get data from a device that uses FTMS, and found the following problem:

According to [the bluetooth assigned numbers repo](https://bitbucket.org/bluetooth-SIG/public/src/4164a40227fd8d3ec4c2dc143f664ce1894e98cf/assigned_numbers/uuids/characteristic_uuids.yaml#lines-614), the fitness machine status characteristic id is `00002ada-0000-1000-8000-00805f9b34fb`.

Currently it has value `00002ad3-0000-1000-8000-00805f9b34fb`, which is the training status characteristic id.

zacharyedwardbull, 2024-01-02:
@tensorturtle would you mind taking a look at this FTMS question? Cheers

tensorturtle, 2024-01-03:
@coniesco Thanks for raising this issue. If memory serves, the `00002ada...` characteristic gives very limited information about the trainer itself, and the `00002ad3...` is the one that provides useful cycling data. If you could specify in greater detail what error message you saw, or what data you're trying to get out, I can help you better.

coniesco, 2024-01-05:
@tensorturtle Sure! The `0002ad3...` characteristic is actually duplicated in the code. It's assigned to [training status here](https://github.com/zacharyedwardbull/pycycling/blob/4bb1f213f4c4c2c3cbf1cdbe5832cd575d4bedfb/pycycling/fitness_machine_service.py#L51), which is correct according to the spec and what you are referring to. But it is also assigned to the [fitness machine status](https://github.com/zacharyedwardbull/pycycling/blob/4bb1f213f4c4c2c3cbf1cdbe5832cd575d4bedfb/pycycling/fitness_machine_service.py#L49), which is the issue I found.

zacharyedwardbull, 2024-01-06:
Released 0.3.4 with the fix.

closed 2024-01-06 by zacharyedwardbull