zacharyedwardbull/pycycling #35: Simulate FTMS BLE device (smart indoor bike)

dfxh, 2024-02-06:
Hi,
I have the following issue:
- an "old" fitness machine (indoor bike) with adjustable power but NO BLE/FTMS interface
-  the request to connect to a smart trainer. e.g. ZWIFT, Wahoo SYSTM, bkool, ..... 

I would need the option of pycycling to simulate a smart bike like wahoo kickr or TACX etc.
e.g. I start pycycling (maybe as a service) and connect to systems via BLE like ZWIFT simulating a real "smart" indoor bike.
I need the option to change information such as power, cadence, heartrate with pycycling via console or some (web interface).

Furthermore with this interface to report power, cadence, heartrate and get the should values of power and cadence within pycycling I can create a interface (via the pi's GPIO pins or some shield) to the "old fitness machine" and set the values there.

OevreFlataeker, 2024-02-06:
Maybe have a look at this project here: https://ftmsemu.github.io/
Guess that's pretty much the same basic idea behind it.
This might act as a starter for your journey: https://github.com/OevreFlataeker/ftms_emu

zacharyedwardbull, 2024-05-29:
@dfxh did you manage to simulate the FTMS device? Please could you update with your findings and then I will close the issue.