zacharyedwardbull/pycycling #36: FTMSControlPointOpCode wrong bit

bytosaur, 2024-02-26:
Dear @zacharyedwardbull,

thank you for building this library! You really helped me out a lot.

I am currently reading the [FTMS BT specification 1.0](https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=423422) (see page 52) and stumbled across a misplaced bit: 

opcode __0x06__ reffers to __SET_HEART_RATE__ and not __START_AND_STOP__, which also affects the op codes afterwards.

BTW: there are many features and settings not implemented and i d like to add them later. Is there a reason for not implementing everything?

zacharyedwardbull, 2024-02-26:
Thanks for the issue/PR @bytosaur 

@tensorturtle  please could you have a look at this, seems to be related to the FTMS implementation.

Cheers,
Zach

tensorturtle, 2024-02-27:
@bytosaur Thanks for submitting this issue. 

It seems that I was following [Huawei's documentation](https://developer.huawei.com/consumer/en/doc/HMSCore-Guides/fmcp-0000001050147089) instead of the [Bluetooth.org standard, page 52](https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=423422). Given the discrepancy, obviously the official docs wins, so I'll look into this further and test it with my indoor trainer. 

Since FTMS covers not only smart trainers but also rower, treadmill, elliptical, and others, you're right that there's a lot of features that aren't implement in this cycling-focused library.
I have attempted to implement all features that can be demonstrated with a fully featured smart trainer such as Elite Suito T.
You may also notice that FTMS sort of expects the master to send specific workout session related information to the slave (indoor trainer), but I'm not familiar with / aware of any devices that actually use that data, so those aren't implemented either. For example, https://github.com/zacharyedwardbull/pycycling/blob/master/pycycling/ftms_parsers/training_status.py is not actually used at all in this library's FTMS example.

I believe PRs are welcome. What devices do you have available for testing?

tensorturtle, 2024-02-27:
@bytosaur I see you already submitted a PR for this fix. Would you like me to test your fork? I think I can make time today

bytosaur, 2024-02-27:
On my fork I implemented the rest of the opcodes.
I have an elite setup with Justo and rizer. 
I just used the simulation command which is needed to control the RIZER through the JUSTO. It works fine, but I did not test the other functions.

Additionally, there is another section to the machine features. It checks what is allowed for writing to. I have added this too and changed the method in the ftms service accordingly.

I am very happy to help :))

tensorturtle, 2024-02-27:
@bytosaur Very nice! I don't have the rizer myself so it's great that you have access to it. 
I think it would be helpful if you modified [examples/fitness_machine_service_example.py](https://github.com/zacharyedwardbull/pycycling/blob/master/examples/fitness_machine_service_example.py) or created a new example for rizer to demonstrate your changes to the code.
And after that if you can run the examples without problems, I can optionally validate it on my hardware before merging.

Thank you for contributing

bytosaur, 2024-02-27:
Unfortunately, the elite setup i have is not mine and will soon be used in a museum so I cannot test too many features. 
So my goal was to use these features: speed (justo),  slope simulation (justo / elite) and steering (elite).
I just got into Bluetooth and i have even less experience with Fitness Trainers, which is why this repository helped me quite a lot <3

I will adjust the [examples/fitness_machine_service_example.py](https://github.com/zacharyedwardbull/pycycling/blob/master/examples/fitness_machine_service_example.py) a little bit to showcase setting the Indoor Bike Simulation Parameters.
I can do a rizer service and example according to a sheet elite send me. Compared to Sterzo there is no challenge file afaik, but a control point for calibration.

zacharyedwardbull, 2024-04-26:
Excellent work thanks for the contribution @bytosaur and thanks for checking @tensorturtle. I've just published a release, 0.4.0, with the changes!

closed 2024-04-26 by zacharyedwardbull