zacharyedwardbull/pycycling #39: Tacx Neo 2 sends message only about 1 time per second

Florian2501, 2024-03-25:
We work with the tacx neo 2 and receive speed values only about one time per second. Are there any ways to increase this? It leads to a massive delay in the simulator.

zacharyedwardbull, 2024-03-25:
Hi Florian, 

May I ask which protocol are you using?

I did a similar simulator project as my university project, and I remember for the Tacx Neo 2 using the Tacx Trainer Control (ANT+ FE-C over BLE) protocol, data was sent at the following intervals:

- 1 Hz for speed/cadence measurements
- 4 Hz for power measurements

We used our own physics model and then used the more frequent power measurements to calculate speed from the physics model. We then did resynchronisation of our model with the trainer every time a new speed measurement came in to prevent large divergence between the two.

I've included a diagram from my project report which illustrates this:

[image]

Cheers,
Zach

Florian2501, 2024-03-25:
Hello Zach,

thank you very much! I use the Bluetooth Low Energy Connection to connect with a Elite Sterzo Smart and the Tacx Neo 2. The connection works good and i get nice smooth values fromthe Elite but the values from the Tacx come only about 1 time per second. Currently I use the speed value to controll my simulated bicycle in Unity but this is with a big delay. So if I understand you correctly, it is not poissbile to get the speed values with more than 1 Hz?
But you used a workaround to calculate the speed based on the power which gets updated more often (4 Hz)? How do I get those power values? Is it with instantaneous_power from set_specific_trainer_data_page_handler?
Thank you!!!
Flo

zacharyedwardbull, 2024-03-25:
Yep `instantaneous_power`. It's a bit of effort to make your own physics model but that should allow you to reduce the latency in your simulation.

Florian2501, 2024-03-26:
Thank you very much!

closed 2024-03-26 by Florian2501