zacharyedwardbull/pycycling #41: Reset Distance Travelled Value in GeneralFEData

NicksIDE7, 2024-04-23:
Hello, anyone not how to reste the distance travelled in GeneralFEData? 

The following code shows the distance travelled, but i cant reset it.

`
import asyncio
import pandas
import sys
from io import StringIO

from bleak import BleakClient

from pycycling.tacx_trainer_control import TacxTrainerControl

async def run(address):
    async with BleakClient(address) as client:
        def my_measurement_handler(data):
            global distance
            distance = data[2]
            print("DISTANCE TRAVELLED : {0} ".format(distance))

        await client.is_connected()
        trainer = TacxTrainerControl(client)
        trainer.set_general_fe_data_page_handler(my_measurement_handler)
        await trainer.enable_fec_notifications()
        await asyncio.sleep(100.0)
        await trainer.disable_fec_notifications()

if __name__ == "__main__":
    import os

    os.environ["PYTHONASYNCIODEBUG"] = str(1)

    device_address = "E7:AB:9F:ED:EA:2B"
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run(device_address))

`

zacharyedwardbull, 2024-04-26:
I'm not sure if there is a way to reset the distance travelled. One way you could work around this is to save whatever the distance travelled is when you want to "reset" it, and then subtract that value from future distance travelled values. You might also need some extra logic to handle wraparound (when the distance travelled goes above whatever the max in on the trainer).

Sorry this answer isn't the best, unfortunately it's a little while since I worked directly with a Tacx trainer.

NicksIDE7, 2024-04-29:
Thank you for the comment. I guess the distance_traveled variable in GeneralFEData is an 8- Bit Value. I tested it and at 256 the value reset to zero. There should be a conversion to uint or float? Or any suggestions which type of storage area the variables have ?

zacharyedwardbull, 2024-04-30:
See this screenshot from the ANT+ FE-C specification for the General FE data page:

[image]

So yep, it's an integer which gives distance in metres.

NicksIDE7, 2024-05-02:
thank you for this answer. you are right, maybe i have to use a counter to set up a flag when the value wraparound or some other logic to fix this issue as you mentioned below.

closed 2024-05-02 by NicksIDE7