zacharyedwardbull/pycycling #42: Bryton Gardia R300L not compatible

OndrejBakan, 2024-04-26:
Hello, similarly to #20, Bryton uses it's own bluetooth characteristic for radar data.

It looks like there's no specific GATT characteristic for radar in the first place, so it's up to manufacturer to choose where to publish the radar data. Bryton uses characteristic `f3641401-00b0-4240-ba50-05ca45bf8abc` and the packet format is very similar, if not equal, to the ANT+ RDR standard.

It's unlucky that there's no common service.

zacharyedwardbull, 2024-04-26:
Hey @OndrejBakan! Thanks for mentioning this, I agree it's a shame it's not standardised like some of the other protocols. Perhaps I should update the README to make it clear that the current Radar class probably only works for Garmin.

As always, you're welcome to implement the protocol for your device and I will review and merge it into the project. Tacx actually has something similar where it uses the ANT FE-C standard over BLE.

OndrejBakan, 2024-04-28:
Hello again @zacharyedwardbull!

Could you, please, take a look at my code before I commit to finish it? Would you be OK with an implementation like this?

https://github.com/OndrejBakan/pycycling/commit/1429e4fb7393502e71cbda8cc039fcc804d0ab5e#diff-a8b9990e2e24f5af1f7e4b6b0b16951471db24c3f58c8ddf0fbe7d71cc61c0f8

zacharyedwardbull, 2024-04-28:
Hey! Thanks for putting this together, looks really nice!

I like the way you've done it, but I think my preference would be to separate it out into a new file/service rather than combine with the Garmin one. My reasons are:

- Unless you have a Garmin radar device we can't test the code easily to check it still works (unfortunately I don't have one either)
- We're changing the number of elements in the `RadarMeasurement` namedtuple, which means if people are referring to elements using indexes then they would get different values, so it's not a backwards compatible change
- Easier to reason about with less abstraction even if it means a little bit of duplicated code

Open to discussion though of course! Also @tensorturtle feel free to share your opinion since you wrote the Garmin support.

Cheers!