zacharyedwardbull/pycycling #46: error when calling set target resistance level

nbazimi, 2024-11-13:
Hi 
I have smart trainer Tacx Neo 3m.
this error occurs when calling fitness machin example: set target resistance level

Traceback (most recent call last)
File "C: \Users \Baran\Desktop \Arbeit\New folder \pycycling-masterlexamples\fitness machine service example.py", line 47, in <module>
loop.run_until_complete(run(device_address))
File "C: \Users\Baran\AppData\LocallPrograms\Python\Python37\liblasyncio\base_events.py", line 579, in run_until_complete
return future.result()
File "C:\Users\Baran\Desktop\Arbeit\New folder\pycycling-master\examples\fitness machine_service example.py", line 25, in run
await ftms.set_target_resistance_Level(10)
File "C: \Users\Baran\Desktop\Arbeit\New folder\pycycling-master\pycyclinglfitness machine service.py", line 235, in set_target_resistance leve
ftms_fitness_machine_control_point_characteristic_id, message, True
File "C:\Users\Baran\Desktop\Arbeit\New folder\pycycling-master\.venv\lib\site-packages\bleakl init _py", line 659, in write_gatt_char
await self._backend write_gatt_char(char_specifier, data, response)
File "C: \Users\Baran\Desktop \Arbeit\New folder\pycycling-masterl.venv\lib\site-packages\bleaklbackends\winrt\client.py", line 876, in write_ge
f"Could not write value (data) to characteristic (characteristic.handle:04X)",
File "C:\Users\Baran\Desktop\Arbeit\New folder\pycyclinq-master\.venv\lib\site-packages\bleak\backends\winrt\client.pỵ", line 116, in _ensure.
f"(fail_msgł: Protocol Error 0xfresult.protocol_error:02X): ferr)"
bleak.exc.BleakError: Could not write value b'Ix04\n' to characteristic 004B: Protocol Error 0xFD: Client Characteristic Configuration Descriptc.
[image]

bytosaur, 2024-11-17:
Hey @nbazimi,

Firstly, I havent tried the system on windows. Any chance to test it on linux or Mac? I tested this on RPi5 and MacM1.

Secondly, can you share the complete file you r running? It looks like you have modified the example.
The line you r referring in your code is 25 while in the original example it is 90.

Thirdy, please copy the whole error message. The message you sent looks like it is missing characters and doesnt make sense, e.g. the file `pycyclinglfitness machine service.py` does not exist in the repository.

tensorturtle, 2024-11-17:
Hi I'm monitoring this issue and will chime in if I see anything I can help with. Thanks @bytosaur for taking initiative to resolve this issue.

nbazimi, 2024-11-18:
> Hey @nbazimi,
>
> Firstly, I havent tried the system on windows. Any chance to test it on linux or Mac? I tested this on RPi5 and MacM1.
>
> Secondly, can you share the complete file you r running? It [...]

this is what i used:

import asyncio
from bleak import BleakClient
from pycycling.fitness_machine_service import FitnessMachineService

async def run(address):
    async with BleakClient(address, timeout=10) as client:
        ftms = FitnessMachineService(client)

        # Print 'read' characteristics

        ### Fitness Machine Features
        fitness_machine_features, target_setting_features = await ftms.get_fitness_machine_feature()
        fitness_machine_features = fitness_machine_features._asdict()
        target_setting_features = target_setting_features._asdict()

        supported_resistance_level_range = (
            await ftms.get_supported_resistance_level_range()
        )
        print("Supported resistance level range:")
        print(supported_resistance_level_range)
        print()
        max_resistance = supported_resistance_level_range.maximum_resistance
        print("Setting target resistance level to 25 percent of maximum resistance level...")
        await ftms.set_target_resistance_level(10)

        await asyncio.sleep(5)

        print("Increasing target resistance level to 50 percent of maximum resistance level...")
        await ftms.set_target_resistance_level(50)

        await asyncio.sleep(5)

        # Reset target resistance level
        print("Resetting target resistance level...")

        await ftms.reset()

if __name__ == "__main__":
    import os

    os.environ["PYTHONASYNCIODEBUG"] = str(1)

    device_address = "E1:11:03:A9:40:9C"
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run(device_address))

    # Unresolved intermittent bug when writing to the control point characteristic:
    # bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Operation failed with ATT error: 0x80 (Unknown code)
    #
    # To work around this, just retry on error, like:
    #
    # import bleak
    # while True:
    #     try:
    #         loop.run_until_complete(run(device_address))
    #     except bleak.exc.BleakDBusError as e:
    #         print("BleakDBusError, retrying...")
    #         print(e)
    #         continue
    #     except KeyboardInterrupt:
    #         break

tensorturtle, 2024-11-19:
@nbazimi 

First, could you first enable DEBUG level logging and then please copy and paste the error (as you did at the beginning).
https://bleak.readthedocs.io/en/latest/troubleshooting.html#enable-logging 

Second, git clone and run this script that I wrote to check that the trainer supports the right operations on each characteristic. Please run the script, find the trainer, and copy the output and paste it here.
https://github.com/tensorturtle/bluetooth_dissect

Third, thank you for pasting the script you've used, but for now can we focus on running the provided example script? That would reduce the confusion.

I have used this FTMS implementation on Windows (not WSL) so it is expected to work, let's try to figure out why it isn't working.

Shogan, 2025-06-11:
Hey there, I seem to have a different characteristic ID for my fitness machine (Elite Zumo) for writing data (to the one in the source code here used for writing (e.g. set resistance level).

```
ftms_fitness_machine_control_point_characteristic_id = (
    "00002ad9-0000-1000-8000-00805f9b34fb"
)
```
For my Zumo's fitness service the characteristic ID details are:

```
=== CHARACTERISTIC 7 ===
        Characteristic UUID:  00002ad9-0000-1000-8000-00805f9b34fb
        Characteristic handle:  49
        Characteristic description:  Fitness Machine Control Point
        Characteristic properties:  ['write', 'indicate']
                === DESCRIPTOR 1 ===
                Descriptor UUID:  00002902-0000-1000-8000-00805f9b34fb
                Descriptor handle:  51
                Descriptor description:  Client Characteristic Configuration
```

Is it possible to override the one in source with mine? Using the default method for setting resistance on my machine doesn't work.

Shogan, 2025-06-15:
Thanks @tensorturtle , it turns out I was misunderstanding the required setup and sequence needed to set resistance level on my device. 

I had to do the following:
- Use the client to use the `write_gatt_descriptor` function with `bytes([0x02, 0x00])` to enable 'indications'
- request control of the device, using `FTMSControlPointOpCode.REQUEST_CONTROL` code via a control command
- Set resistance level with `FTMSControlPointOpCode.SET_TARGET_RESISTANCE_LEVEL`

So it seems the characteristic uuid was just fine as-is.

tensorturtle, 2025-06-15:
@Shogan Thanks for the investigation. Before we close this issue, is there anything I can improve to prevent such misunderstandings in the future?

Shogan, 2025-06-18:
> [@Shogan](https://github.com/Shogan) Thanks for the investigation. Before we close this issue, is there anything I can improve to prevent such misunderstandings in the future?

sorry for the delay. The only thing for me was understanding the protocol better. Maybe a link to some online documentation that helps users understand more about commands and setting up fitness equipment to accept commands? This bluetooth PDF on this link seems to do the job: https://www.bluetooth.com/specifications/specs/fitness-machine-service-1-0/

tensorturtle, 2025-06-19:
Great idea, thanks for the link! @zacharyedwardbull If you can please update the docs with the link and close this issue :)

zacharyedwardbull, 2025-06-24:
Thanks @tensorturtle, the documentation is already linked in the README, closing this issue now.